ARM-software/SDS-Framework #57: Check dependency on include files

ReinhardKeil, 2025-02-25:
Check the implementation for unused header files and remove them. These create potentially dependencies on compilers (i.e. AC6).

See this:

[image]

ReinhardKeil, 2025-05-13:
@DavidLesnjak has this been done already during the development?

DavidLesnjak, 2025-05-13:
Partly, I'll go through all modules again.

DavidLesnjak, 2025-06-13:
Reviewed SDS source files for unused or problematic headers.

The `stdatomic.h` header requires the `C11` standard. To improve portability and maintainability I added a conditional guard to include it only when supported. See [Commit e46d42a](https://github.com/ARM-software/SDS-Framework/commit/e46d42a7b464b003ffcb82e240ae2b4477ac18c0)

closed 2025-06-13 by ReinhardKeil