ARM-software/SDS-Framework #59: Consider to add a `header:` in addition to `content:` in schema

ReinhardKeil, 2025-02-25:
To capture a header before a data stream that is repeated consider to add `header:` to the schema.
https://github.com/ARM-software/SDS-Framework/tree/main/schema

Example:

```
struct { 
  uint32_t info1                 // header: info1 + info2
  uint32_t info2.
  struct {                           // content: sensor data stream format
    uint16_t x;
    uint16_t y;
    uint16_t z;
  } accelerometer [30];             // number of samples in one data stream record
}
```

```yml
sds:
  header:
    - value: info1
      type: uint32_t
    - value: info2
      type: uint32_t
  content:
    - value: x
      type: uint16_t
    - value: y
      type: uint16_t
    - value: z
      type: uint16_t
```

ReinhardKeil, 2025-11-05:
> @ReinhardKeil<https://github.com/ReinhardKeil> ,
>
> it seems that the example you provided can already be implemented today by simply sending a data block that contains only useful data - i.e., not sending a full-sized block if it doesn’t contain useful content.
>
> In the current implementation, the block size<https://github.com/ARM-software/SDS-Framework/blob/54ffed9e8a111c698146ebf52b4a82c81088f151/sds/source/sds_rec_play.c#L40> already implicitly represents the number of items in the data block.
>
> Introducing a header would require updating several utilities (SDS-View, SDS-Convert, and SDS-Check) to process this additional header information and to interpret its context correctly.
>
> Do you think it’s worth investing time in developing this feature?
>
>
> IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

Milorad,

We first need to update the Alif pack before we work on MLEK and SDS examples. In this context could you review the USB driver and check if the problems are resolved.

Reinhard

ReinhardKeil, 2026-10-06:
is this feature required?  We should investigate some object detection use-cases

MiloradCvjetkovic, 2026-10-06:
Hi @ReinhardKeil , 

if I understand correctly, the intention was to address a variable number of records within a data block.

However, this functionality is already covered by the specification: https://arm-software.github.io/SDS-Framework/main/theory.html#sds-metadata-format, under `Data blocks, records, and values`. Therefore, I don’t think it needs to be introduced separately in the way proposed here.

IMHO, this feature is no longer required.

ReinhardKeil, 2026-11-09:
Feature is required. You have seen examples

ReinhardKeil, 2026-11-09:
For SDS data formatted with header, JSON is a suitable output format. Each line represents a complete record, including its header information and a variable-length array. 

For example:
```json
{"time":0.000,"count":2,"content":[{"x":120,"y":80,"w":40,"h":90},{"x":300,"y":100,"w":35,"h":85}]}
{"time":0.033,"count":1,"content":[{"x":125,"y":82,"w":40,"h":90}]}
{"time":0.066,"count":0,"content":[]}
```