ARM-software/SDS-Framework #60: Questions on SDSIO-Convert

ReinhardKeil, 2025-02-26:
- What is the metadata file required for WAV files. Do we have an example?
- The start-tick and stop-tick argument is a value in which unit (I assume in seconds).  Does it assume that tick is a 1ms interval.  Does the value consider the tick-frequency in the metadata file?

MiloradCvjetkovic, 2025-03-03:
I think this is an example of metadata file for audio data https://github.com/ARM-software/SDS-Framework/blob/main/examples/recordings/Microphone.sds.yml

MiloradCvjetkovic, 2025-04-25:
This was addressed and current documentation should cover that.

ReinhardKeil, 2025-05-13:
The documentation for WAV is still weak: https://arm-software.github.io/SDS-Framework/main/utilities.html#audio-wav

- What is the relationship to the WAV file header https://en.wikipedia.org/wiki/WAV
- What is the audio format used?
- How is the data type converted?

MiloradCvjetkovic, 2025-05-21:
This was addressed by @TeoMahnic in 70e9ce22ab10f622cabdc74ea2792ce20db672a9 .

closed 2025-05-21 by ReinhardKeil