ARM-software/SDS-Framework #62: Question: how is <index> controlled?

ReinhardKeil, 2025-02-26:
The SDS data stream is recorded to files with the following naming convention:

```
<name>.<index>.sds
```

`<name>`  is a parameter in `sdsRecOpen` or `sdsPlayOpen`.  

But how is `<index>` controlled? Is each `sdsRecClose` or `sdsPlayClose` incrementing it?

DavidLesnjak, 2025-03-04:
For recording, the indexing mechanism works as follows:

- sdsRecOpen internally calls sdsioOpen (for write) to handle file creation.
- sdsioOpen creates a new file with the format: <name>.<index>.sds.
- The name parameter is passed from sdsRecOpen to sdsioOpen.
- The index starts at 0 and is automatically incremented for each new sdsioOpen call with the same name parameter.
- Before creating a new file, sdsioOpen checks if a file with the requested <name>.<index>.sds already exists:
-- If the file exists, the index is incremented
-- This process continues until the first available (unused) index is found.

ReinhardKeil, 2025-03-07:
Covered on SDSIO_CMD_OPEN in 
https://arm-software.github.io/SDS-Framework/main/theory/#sdsio-server-protocol

ReinhardKeil, 2025-04-09:
we consider this as complete. Maybe add some more to the documentation

closed 2025-04-11 by ReinhardKeil