ARM-software/SDS-Framework #68: Create sequence diagrams

ReinhardKeil, 2025-03-03:
For the section https://arm-software.github.io/SDS-Framework/main/theory/.

1. There should be sequence diagrams that explain how the sdsio operations are protected for multi-threading.
 
2. Clear usage instructions for multi-threading applications.
   - can sdsclose called in the middle of a sdswrite/read operation?
   - should sdsopen/write/read executed from the same thread?

ReinhardKeil, 2025-03-05:
SDSIO Server commands are covered here https://github.com/ARM-software/SDS-Framework/blob/main/sds/source/sdsio/serial/usart/sdsio_serial.c

ReinhardKeil, 2025-06-04:
@DavidLesnjak I believe this is complete.  If so, please close it.

MiloradCvjetkovic, 2025-06-12:
@ReinhardKeil ,

I do not see that all requirements of this issue were addressed but if you think they were, as you are the creator of the issue, please close it.