ARM-software/SDS-Framework #76: sdsioOpen NULL in sdsRecOpen

MaxCYCHEN, 2025-03-13:
Hi,
Thanks for your SDS_Framework project.
May I ask about `sdsRecId_t sdsRecOpen()`? I tried your UART example.
If in recorder mode, the rec->sdsio will be set to `0` by `sdsioOpen(name, sdsioModeWrite)` which set return value `sdsio_id = 0U`. 
This will cause the latter judgement to clear the stream obj and let the following Recorder write failed.
`if ((rec->stream == NULL) || (rec->sdsio == NULL)) {
        if (rec->stream != NULL) {
          sdsClose(rec->stream);
          rec->stream = NULL;
        }
        if (rec->sdsio != NULL) {
          sdsioClose(rec->sdsio);
          rec->sdsio = NULL;
        }
        sdsRecFree(index);
        rec = NULL;
      }`

Because I ran this code in our own board and encountered this issue, or this is my misunderstanding?

Thank you !

MiloradCvjetkovic, 2025-03-13:
Hi @MaxCYCHEN , 

can you just provide a short conclusion of what you found out please?

Was it that SDSIO-Server was not running, or something else?

MaxCYCHEN, 2025-03-13:
Hi @MiloradCvjetkovic,
Sorry for closing it directly.
Yes, I had opened SDSIO-Server.py, but it looks like `sdsioReceive(&header, NULL, 0U)` didn't receive any value.
But in SDSIO-Server.py, it actually accepted the data. I still checking if I need to change something in SDSIO-Server.py.

**Board printf:**
First send in sdsioOpen()
pDrvUSART->Send header: 16, data: 14

**SDSIO-Server.py print**
Receive len:

Server opening...

  Serial Port: COM8

Server Opened.

Request buffer size: 3600, Request buffer count: 16

Request buffer size: 3600, Request buffer count: 31

closed 2025-03-14 by jkrech