ARM-software/SDS-Framework #87: Generate SDS control module

ReinhardKeil, 2025-03-15:
Suggest to create a generic "sds_control.c" module based on https://github.com/Arm-Examples/SDS-Examples/blob/main/Hardware/DataTest/rec_management.c

Modifications:
 
- https://github.com/Arm-Examples/SDS-Examples/blob/main/Hardware/DataTest/rec_management.c#L90
     - idle counter can be evaluated in just 10msec.
     - this removes the need for /100 in https://github.com/Arm-Examples/SDS-Examples/blob/main/Hardware/DataTest/rec_management.c#L164

**State Machine**
- introduce a state machine with states
   - SDSIO_CLOSED (0) - No SDS I/O pipes open.
   - SDSIO_OPEN (1) - SDS I/O pipes are open and ready for read/write operations.
   - SDSIO_CLOSING (2) – Indicates the transition toward the closed state, complete current pending algorithm execution.
   - SDSIO_HALTED – Indicates a completed algorithm execution after SDSIO_CLOSING state.  
  
- https://github.com/Arm-Examples/SDS-Examples/blob/main/Hardware/DataTest/rec_management.c#L57
   - should become 'sdsio_state'.

- This https://github.com/Arm-Examples/SDS-Examples/blob/main/Hardware/DataTest/rec_management.c#L142 becomes SDSIO_OPEN.

- This https://github.com/Arm-Examples/SDS-Examples/blob/main/Hardware/DataTest/rec_management.c#L144 becomes SDSIO_CLOSING.

- State SDSIO_HALTED becomes active when state is SDSIO_CLOSING and test data are written https://github.com/Arm-Examples/SDS-Examples/blob/main/Hardware/DataTest/DataTest.c#L102

- One state SDSIO_HALTED this is executed: https://github.com/Arm-Examples/SDS-Examples/blob/main/Hardware/DataTest/rec_management.c#L146

**Allow to measure Jitters**
Use a separate timestamp value here 
https://github.com/Arm-Examples/SDS-Examples/blob/main/Hardware/DataTest/DataTest.c#L102

ReinhardKeil, 2025-03-24:
I believe jitter should be measured using a data validation utility (not in the application itself).  I would therefore remove it,

Checking the state in a switch/case has the draw-back that it may take another interval until the end of recording is captured.

Hence rewrite the code as follows.

```c
  interval_time = osKernelGetTickCount();

  for (;;) {
    if (sdsio_state == SDSIO_OPEN) {
        timestamp = osKernelGetTickCount();
        CreateTestData();

        n = sdsRecWrite(recIdDataInput, timestamp, &imu_buf, sizeof(imu_buf));
        SDS_ASSERT(n == sizeof(imu_buf));

        n = sdsRecWrite(recIdDataOutput, timestamp, &ml_buf, sizeof(ml_buf));
        SDS_ASSERT(n == sizeof(ml_buf));
     }

    if (sdsio_state == SDSIO_CLOSING) {
        // Algorithm execution stopped, transit to halted state
        sdsio_state = SDSIO_HALTED;
    }

    interval_time += 10U;
    osDelayUntil(interval_time);
  }
```

furbanc, 2025-03-25:
Most of the proposed changes have now been incorporated.

The creation of `AlgorithmThread` is not moved to the `app_main` function, as this affects the evaluation of the initial value of the `no_load_cnt` counter.

ReinhardKeil, 2025-03-26:
closing this ticket.  Maybe there is a better way to get to the `no_load_cnt` value, but for now it is OK

closed 2025-03-26 by ReinhardKeil