ARM-software/SDS-Framework #89: Does it make sense to combine Record and Play functionality?

ReinhardKeil, 2025-03-17:
Does it make sense to combine Record and Play functionality?

It reduces user complexity and configuration options.  sds_config.h/sds_play_config.h/sds_rec_config.h can be consolidated. Typically SDS:Buffer is not used without SDS:RecPlay for we could keep this separate for now.

This Component View:

[image]

Could change to:

Component | Description
:-----------------|:--------------------------
SDS:Buffer         | 
SDS:RecPlay       | Recorder/Player interface to algorithm under test

MiloradCvjetkovic, 2025-03-17:
For discussion, current API of both components:

| Recorder       | Player
| -------------- | -----------
| sdsRecInit     | sdsPlayInit
| sdsRecUninit | sdsPlayUninit
| sdsRecOpen   | sdsPlayOpen
| sdsRecClose  | sdsPlayClose
| sdsRecWrite  |
|                       | sdsPlayRead
|                       | sdsPlayGetSize
|                       | sdsPlayEndOfStream

ReinhardKeil, 2025-03-17:
We agree to keep the names as suggested in implement this.

DavidLesnjak, 2025-03-25:
@ReinhardKeil, In [develop](https://github.com/ARM-software/SDS-Framework/tree/develop) branch I prepared [commit](https://github.com/ARM-software/SDS-Framework/commit/1968fe4a790f1b6cb08ba0fef0ebded61a8faa15), which address this issue. Please take a look before I merge it to main branch.

As agreed, sds_rec and sds_play are combined into one driver sds_rec_play. Consequentially there is now one SDS:RECPLAY component. 

Recorder and player now share the same:
- configuration file: sds_rec_play_config.h
-  handle type: sdsRecPlay_t
-  temporary buffer: RecPlayBuf
-  thread: sdsRecPlayThread
-  functions: sdsRecPlayInit and sdsRecPlayUninit

DavidLesnjak, 2025-03-26:
Implemented in this [commit](https://github.com/ARM-software/SDS-Framework/commit/81af14bfc2e30e23cf50b18794ce04785ec19506)

DavidLesnjak, 2025-03-31:
I did a major update of the sds_rec_play driver - [commit](https://github.com/ARM-software/SDS-Framework/commit/688f5cd4ba293238a9d7f6717c6a2075d52d4be1).

Main changes:

- sdsPlayOpen pre-fills SDS Stream Buffer
- Added timeouts for Open and Close operations
- Introduced and tracked initialization flag
- Introduced and managed stream states: opening, opened, closing, rw_active
- Enhanced inline comments

ReinhardKeil, 2025-04-14:
this step is complete, however I have several review comments in new tickets.

closed 2025-04-14 by ReinhardKeil