ARM-software/SDS-Framework #91: Error Codes for Timeout, Overflow, Underflow

ReinhardKeil, 2025-03-17:
I would expect error codes for

timeout (for Open/Close functions) - when Server does not respond in a given time.  I guess this could also occur if the connection breaks down.

overflow (for RecWrite function) - when Recorder does not have space in SDS buffer

underflow (for PlayRead function) - when Player has not provided data in SDS buffer

DavidLesnjak, 2025-04-02:
1. #define SDS_REC_PLAY_ERROR_TIMEOUT (-2) has been added to sds_rec_play.h to handle timeout scenarios (e.g., when the server does not respond during Open/Close). The actual timeout detection and handling logic still needs to be implemented.

2. Overflow handling: The sdsRecWrite function is designed to return the number of bytes actually written to the SDS circular buffer. It does not overflow — if the buffer is full or partially full, it simply writes what it can and returns the number of bytes successfully written. So there's no need for a dedicated overflow error code.

3. Underflow handling: Similar to sdsRecWrite, the sdsPlayRead function returns the number of bytes actually read from the SDS buffer. If there isn't enough data available, it reads as much as it can. Therefore, underflow is also not treated as an error and probably doesn't require a specific error code.

ReinhardKeil, 2025-04-17:
Instead of Overflow use RecOutError
Instead of Underflow use PlayInError

DavidLesnjak, 2025-05-08:
We've reconsidered the situation and updated the API accordingly.
New return codes and event definitions have been introduced in the following commit: [Commit d76f7ab](https://github.com/ARM-software/SDS-Framework/commit/d76f7ab199572511f5838a8ffc42dcfe052fd8b1#diff-f878cb693a5c2b0973a8f24a1611bf09616999c57f1cfb8a486ec0ab197212a1R242)

```
// Function return codes
#define SDS_REC_PLAY_OK                 (0)     // Operation completed successfully
#define SDS_REC_PLAY_ERROR              (-1)    // Operation failed
#define SDS_REC_PLAY_ERROR_PARAMETER    (-2)    // Operation failed: parameter error
#define SDS_REC_PLAY_ERROR_TIMEOUT      (-3)    // Operation failed: timeout error
#define SDS_REC_PLAY_ERROR_IO           (-4)    // Operation failed: SDS I/O interface error
#define SDS_REC_ERROR_NO_SPACE          (-5)    // Operation failed: insufficient space in stream buffer
#define SDS_PLAY_ERROR_NO_DATA          (-6)    // Operation failed: insufficient data in stream buffer
#define SDS_PLAY_EOS                    (-7)    // End of stream reached

// Event codes
#define SDS_REC_PLAY_EVENT_ERROR_IO     (1UL)   // SDS I/O interface error
#define SDS_REC_EVENT_ERROR_NO_SPACE    (2UL)   // sdsRecWrite() failed: insufficient space in stream buffer
#define SDS_PLAY_EVENT_ERROR_NO_DATA    (3UL)   // sdsPlayRead() failed: insufficient data in stream buffer
```

closed 2025-05-13 by ReinhardKeil