ARM-software/SDS-Framework #94: Python utility for verifying test data

ReinhardKeil, 2025-03-17:
Create another utility here https://github.com/ARM-software/SDS-Framework/tree/main/utilities

Suggested name:  SDS-Check

This should verify the SDS data files that are written with:
https://github.com/Arm-Examples/SDS-Examples/tree/main/Hardware/DataTest

Test aspects:
  - correctness of data transmission
  - jitter of the recording interval

See https://github.com/ARM-software/SDS-Framework/issues/63

furbanc, 2025-04-08:
Updated, pull request created [cba3d4](https://github.com/ARM-software/SDS-Framework/pull/100/commits/cba3d4c5179e59ffec4f1dafb66199bcb606463e)

ReinhardKeil, 2025-04-09:
Print statistics:
  - Datarate per tick
  - Total number of records
  - Min/Av/Max size in bytes

furbanc, 2025-04-10:
Requested changes implemented, pull request [created](https://github.com/ARM-software/SDS-Framework/pull/101)

ReinhardKeil, 2025-04-14:
completed with https://github.com/ARM-software/SDS-Framework/commit/24e4618a4c5f6d5a723ac18c79cbb8cb080fe3fa

closed 2025-04-14 by ReinhardKeil