ARM-software/SDS-Framework #96: Move Common SDSIO Client Functions to a Dedicated Driver

DavidLesnjak, 2025-03-24:
Currently, each SDSIO implementation that communicates with the SDSIO-Server contains identical implementations of the following functions:
- `sdsioOpen`
- `sdsioClose`
- `sdsioWrite`
- `sdsioRead`
- `sdsioEndOStream`

These functions only differ in their use of low-level I/O operations (e.g., sdsioSend and sdsioReceive), which vary between transport layers like sockets, USART, or VCOM.

To improve code maintainability and reduce duplication, it would make sense to introduce a new module: `sdsio_client.c`. This module would encapsulate the common client-side logic, while delegating low-level I/O to specific implementations (`sdsio_client_socket`, `sdsio_client_vcom`, `sdsio_client_serial`).

DavidLesnjak, 2025-03-25:
@ReinhardKeil, In [develop](https://github.com/ARM-software/SDS-Framework/tree/develop) branch I prepared [commit](https://github.com/ARM-software/SDS-Framework/commit/9520120e615df4909124fd5aeeebe2ce739507a5), which address this issue. Please take as look before I merge it to main branch.

Note that folder structure of sds/sdsio folder is changed and sdsio component are replaced:

- SDS:IO:Socket                        -> SDS:IO:Client.Socket
- SDS:IO:VCOM.MDK USB        -> SDS:IO:Client.VCOM (MDK USB)
- SDS:IO:Serial.CMSIS USART  -> SDS:IO:Client.Serial (CMSIS USART)

DavidLesnjak, 2025-03-26:
Implemented in this [commit](https://github.com/ARM-software/SDS-Framework/commit/5654c8ac20c5a25ca907c954f2173a2f63657d45) - (including last feedback from @ReinhardKeil)

ReinhardKeil, 2025-04-14:
this is done