ARM-software/SDS-Framework #102: Simplify 'rec_play->state` in sds_rec_play

ReinhardKeil, 2025-04-14:
IMHO the `rec_play->state` should be a simple variable that can have the following values:

#define SDS_RecPlayState_Inactive  0      // data stream unused
#define SDS_RecPlayState_Opening    1    // open function is processed
#define SDS_RecPlayState_ReadActive  2    // data stream is in read mode
#define SDS_RecPlayState_WriteActive 3    // data stream is in write mode
#define SDS_RecPlayState_Closing  4    // close function is processed

Using this states should make:

- simplify the overall code (it can be only one value, not multiple flags).
- `stream_type` obsolete (such a variable should be named mode for consistency).

DavidLesnjak, 2025-04-14:
1. 

While a single state variable may seem simpler, the current design — where stream state is packed into individual bytes — was intentionally chosen to **enable atomic byte access**.

This design is used to cover multithreaded scenarios like the following, where different threads (e.g., writer and closer) need to safely coordinate without interfering with each other's state updates.  
If a single variable (e.g., an enum or bitfield) were used instead, it would require a **locking mechanism** to prevent race conditions during concurrent read-modify-write operations. 

### Example: `sdsRecClose` starts, then `sdsRecWrite` is called shortly after

| Time | Thread      | Action                           | `closing` | `rw_active` | Effect                                                       |
|------|-------------|----------------------------------|-----------|-------------|--------------------------------------------------------------|
| T0   | Thread A    | `sdsRecClose()` is called        | → 1       | 0           | Signals stream is closing                                    |
| T1   | Thread A    | Waits for `rw_active == 0`       | 1         | 0           | Proceeding to notify background thread for final flush       |
| T2   | Thread B    | `sdsRecWrite()` is called        | 1         | → 1         | Sets `rw_active = 1`, enters write                           |
| T3   | Thread B    | Checks `closing == 1`            | 1         | 1           | **Write aborted** due to closing state                       |
| T4   | Thread B    | Clears `rw_active = 0`           | 1         | → 0         | Write safely exits without corrupting stream state           |
| T5   | Thread A    | Sees `rw_active == 0`, continues  | 1         | 0           | Safe to proceed with closing logic (flush buffer, etc.)      |

I suggest keeping current implementation with states packed into individual bytes.

2. 
I agree with changing `stream_type` to `mode`

ReinhardKeil, 2025-04-14:
Thanks for explaining.

Sorry, I still believe we need to review this code one more time.
A while loop like here https://github.com/ARM-software/SDS-Framework/blob/main/sds/source/sds_rec_play.c#L515
can be toxic when the thread that calls sdsRecClose has a higher priority then the thread that calls sdsRecWrite.

DavidLesnjak, 2025-04-28:
I revised the code and prepared an implementation with a locking mechanism and simplified state handling.
The current implementation of the lock function sdsRecPlayLockAcquire uses the already existing driver functions (load/store exclusive instructions) and could easily be replaced with a mutex if needed in the future.
From a performance perspective, and considering that load/store exclusive is already used in the driver, this currently seems to be the better option.
@furbanc  is already reviewing the code and has performed some initial testing.

Implementation is done in [Commit 027bac5
](https://github.com/ARM-software/SDS-Framework/commit/027bac5e0cfb00af3f70c8cc1a45a91c2bbca665)

closed 2025-05-13 by ReinhardKeil