ARM-software/SDS-Framework #103: Develop and add "Reference Applications" to SDS Software Pack

jkrech, 2025-04-14:
Similar to [**Reference Applications**](https://open-cmsis-pack.github.io/cmsis-toolbox/pack-tools/#reference-applications) in the Keil::MDK-Middleware Software Pack there should be application layers shipped as part of the SDS pack. Ideally available Board Layers working with MDK-Middleware are compatible with SDS application layers.

ReinhardKeil, 2025-04-17:
This is a starting point for a **Reference Applications**:

Three solution projects that differ on the communication technology used for SDS:
  - SDS-Network.csolution.yml
  - SDS-USB.csolution.yml
  - SDS-FileSystem.csolution.yml

Ideally this projects only differ in the communication layer that is used.

Each solution contains:
  - RecordTestData.cproject.yml     # create SDS test files - used to verify bandwidth (also records the results)
  - PlayTestData.cproject.yml          # playback SDS test files - used that a playback works
  - RecordAlgorithm.yml                 # is a template to add user algorithms
  - PlayAlgorithm.yml

IMHO RecordTestData and PlayTestData could be hightly identical, maybe just use a define to switch between the modes.

https://github.com/Arm-Examples/sds-examples  shows how this reference applications are applied to real projects.
Here we show how a EdgeImpulse application works.

jkrech, 2025-05-19:
@ReinhardKeil is this issue superseded by #115 and can be closed?

closed 2025-05-19 by ReinhardKeil