ARM-software/SDS-Framework #112: Add WinUSB mode to SDSIO Server and Examples

ReinhardKeil, 2025-05-13:
There is a bandwidth issue with the current USB implementation which uses VCOM.

Therefore replace VCOM with WinUSB bulk transfer.

MiloradCvjetkovic, 2025-05-13:
We have to investigate this a bit more before going into the implementation.
We should investigate the level of support for WinUSB on Linux and MAC.
If CDC is problematic only on Windows we could see if the CDC device can be accessed on Windows directly from Python instead of as a serial device using Windows serial driver with Python packages like pyusb, pywinusb or WinUsbCDC.

DavidLesnjak, 2025-06-05:
Transfer via USB VCOM has been replaced with USB Bulk transfer:
- Extend SDSIO-Server with USB Bulk [(Commit 440ee69)](https://github.com/ARM-software/SDS-Framework/commit/440ee690787ad69586b0d5d50a9bda6ef99c2c5c)
- Add USB Client driver -using MDK USB Custom Class ([(Commit 881a84e)](https://github.com/ARM-software/SDS-Framework/commit/881a84e1c780590a9a1cf2864d42c3756810d078))
- Update SDS USB layer to use USB Client instead of VCOM client ([Commit db0c413](https://github.com/ARM-software/SDS-Framework/commit/db0c413c19ab25af2f13d3a998ec38624e044f41))
- Remove VCOM Client driver ([Commit 3ddebf6](https://github.com/ARM-software/SDS-Framework/commit/3ddebf6b671c0ef3a8346f1366592fe64aca0584))

closed 2025-06-05 by ReinhardKeil