ARM-software/SDS-Framework #118: Improve AVH Board Layers for Corstone-300

ReinhardKeil, 2025-05-22:
I believe we should only register header files that are user relevant (configure some options). The result is this file 

[Board.clayer.yml.txt](https://github.com/user-attachments/files/20380724/Board.clayer.yml.txt)

I suggest to make the `vsi` directory structure simpler with just:
```
vsi/include
vsi/source
vsi/python
vsi/README.md   # briefly explain the content
```

MiloradCvjetkovic, 2025-05-26:
Addressed in 95d5cb7a996a74ae0ae4880629017a99f75fd943 .

furbanc, 2025-06-16:
VSI support added in [8c7e329b](https://github.com/ARM-software/SDS-Framework/commit/8c7e329b4f93677111e9bf3524ba74e12f72946f)

ReinhardKeil, 2025-06-29:
Layers not fully completed. However it is not in the pack - hence can be done later