ARM-software/SDS-Framework #126: Small Improvement

ReinhardKeil, 2025-06-05:
Not sure if this is easy or not, as Python has it's own way to create help output.

Instead of 
```txt
>python sdsio-server.py
usage: sdsio-server.py [-h] [--verbose] {socket,serial,usb} ...
sdsio-server.py: error: the following arguments are required: server_type
```

Output
```txt
>python sdsio-server.py
usage: sdsio-server.py [-h] [--verbose] {socket | serial | usb} [options]
sdsio-server.py: error: server_type specification is required: {socket | serial | usb}
```

Command line syntax conventions are here
https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/command-line-syntax-key

DavidLesnjak, 2025-06-05:
Currently, all our Python scripts in SDS rely on the default behavior of argparse. To achieve the desired format for the help and error output, we would need to override the default ArgumentParser.error() method.

However, rather than applying this as a one-off solution, it would make more sense to implement this in a generic and reusable way, so it can be shared across all our Python tools.

To do this properly it would likely take a few days of work. Given the current priorities, I don’t think this is feasible to include in the 2.0.0 release.

ReinhardKeil, 2025-08-26:
this is too complicated

ReinhardKeil, 2025-08-26:
Maybe we can somewhat streamline the help screen.  I did not realize that the tool had a workdir as the help screen is incomplete when you enter for example
```
sdsio-server -h usb
```