ARM-software/SDS-Framework #137: Running SDS on AVH-FVP

ReinhardKeil, 2025-08-25:
I try to run the `SDS-Examples\Alif\AppKit\SDS.csolution.yml` with this configuration on AVH-FVP

[image]

This fails with: 

```
Executing task: FVP_Corstone_SSE-300_Ethos-U55 -f Board/Corstone-300/fvp_config.txt c:\w\SDS-Examples\Alif\AppKit\out\DataTest\AVH-SSE-300\DebugRec\DataTest.axf 

Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = 'c:\users\reikei01\appdata\roaming\python\python313\site-packages;'
  program name = 'python'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = 'c:\\Users\\reikei01\\.vcpkg\\artifacts\\2139c4c6\\models.arm.avh.fvp\\11.29.27\\bin\\models\\FVP_Corstone_SSE-300_Ethos-U55.exe'
  sys.base_prefix = 'c:\\Users\\reikei01\\.vcpkg\\artifacts\\2139c4c6\\models.arm.avh.fvp\\11.29.27\\python'
  sys.base_exec_prefix = 'c:\\Users\\reikei01\\.vcpkg\\artifacts\\2139c4c6\\models.arm.avh.fvp\\11.29.27\\python'
  sys.platlibdir = 'lib'
  sys.executable = 'c:\\Users\\reikei01\\.vcpkg\\artifacts\\2139c4c6\\models.arm.avh.fvp\\11.29.27\\bin\\models\\FVP_Corstone_SSE-300_Ethos-U55.exe'
  sys.prefix = 'c:\\Users\\reikei01\\.vcpkg\\artifacts\\2139c4c6\\models.arm.avh.fvp\\11.29.27\\python'
  sys.exec_prefix = 'c:\\Users\\reikei01\\.vcpkg\\artifacts\\2139c4c6\\models.arm.avh.fvp\\11.29.27\\python'
  sys.path = [
    'c:\\users\\reikei01\\appdata\\roaming\\python\\python313\\site-packages',
    '',
    'c:\\Users\\reikei01\\.vcpkg\\artifacts\\2139c4c6\\models.arm.avh.fvp\\11.29.27\\python\\python39.zip',
    'c:\\Users\\reikei01\\.vcpkg\\artifacts\\2139c4c6\\models.arm.avh.fvp\\11.29.27\\bin\\models',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00001688 (most recent call first):
<no Python frame>

 *  The terminal process "C:\WINDOWS\System32\cmd.exe /d /c "FVP_Corstone_SSE-300_Ethos-U55 -f Board/Corstone-300/fvp_config.txt c:\w\SDS-Examples\Alif\AppKit\out\DataTest\AVH-SSE-300\DebugRec\DataTest.axf"" terminated with exit code: 1. 
```

What can be wrong?

furbanc, 2025-08-26:
I think you need a fixed version of Python v3.9  (yours is 3.13)

TeoMahnic, 2025-08-26:
Hi @ReinhardKeil,

The module **`cv2` (OpenCV)** is required to run the VSI Video Server.

To use VSI Video, please ensure the following:
1. Install **Python 3.9 or later** on your PC.
2. Install the missing module with:
   ```sh
   pip install opencv-python
   ```

ReinhardKeil, 2025-08-26:
@jkrech is there an elegant way to get this into the FVP delivery?  Having to fiddle with an old Python installation is not the best UX.

closed 2026-03-14 by ReinhardKeil