ARM-software/SDS-Framework #144: Need SDS file examples for testing

ReinhardKeil, 2025-08-27:
Create for this example https://github.com/Arm-Examples/SDS-Examples/tree/main/ST/B-U585I-IOT02A
several meaningful SDS files that capture motions for the ML model (10 files should be OK)

Store these files in https://github.com/Arm-Examples/SDS-Examples/tree/main/ST/B-U585I-IOT02A/sds_files
And create a README.txt that explains the motion that each file represents.

MiloradCvjetkovic, 2025-08-27:
Currently, 1 recording of input and output data already exists here : https://github.com/Arm-Examples/SDS-Examples/tree/main/ST/B-U585I-IOT02A/algorithm/SDS%20Recordings .
Should I just make 10 new recordings and put them into this existing folder , and remove the existing recordings ?

MiloradCvjetkovic, 2025-08-27:
I already made them now, they are added in https://github.com/Arm-Examples/SDS-Examples/tree/main/ST/B-U585I-IOT02A/algorithm/SDS%20Recordings .